geerlingguy/turing-pi-cluster #21: Drupal image not working

mok0, 2020-06-08:
I wrote this in a comment under your YouTube video, but am not sure if you've seen it.

I am working through your tutorial, and found that I cannot create the Drupal website. It appears to be a problem with Drupal. After choosing the language and then a profile, no matter which profile I choose, I get this error message:

```
Error
The website encountered an unexpected error. Please try again later.
Drupal\Core\Config\UnmetDependenciesException: Configuration objects provided by <em class="placeholder">standard</em> have unmet dependencies: <em class="placeholder">block.block.bartik_account_menu (block), block.block.bartik_help (help, block), block.block.bartik_search (search, block), block_content.type.basic (block_content), comment.type.comment (comment), contact.form.feedback (contact), core.entity_form_display.block_content.basic.default (text), core.entity_form_display.node.article.default (image.style.thumbnail, comment, image, path, text), core.entity_form_display.node.page.default (path, text), core.entity_form_display.user.user.default (image.style.thumbnail, image), core.entity_view_display.node.article.default (image.style.large, comment, image, text), core.entity_view_display.node.article.rss (core.entity_view_mode.node.rss), core.entity_view_display.node.article.teaser (core.entity_view_mode.node.teaser, image.style.medium, image, text), core.entity_view_display.node.page.teaser (core.entity_view_mode.node.teaser, text), editor.editor.basic_html (ckeditor, editor), field.field.block_content.basic.body (field.storage.block_content.body, text, field), field.field.comment.comment.comment_body (field.storage.comment.comment_body, text, field), field.field.node.article.body (field.storage.node.body, text, field), field.field.node.article.comment (comment, field), field.field.node.article.field_image (image, field), field.field.node.article.field_tags (field), field.storage.node.comment (comment, node, field), field.storage.node.field_image (file, image, node, field), field.storage.node.field_tags (node, taxonomy, field), field.storage.user.user_picture (file, image, field), filter.format.basic_html (editor, filter), filter.format.restricted_html (filter), node.type.article (node), rdf.mapping.comment.comment (comment, rdf), rdf.mapping.node.article (node, rdf), rdf.mapping.taxonomy_term.tags (taxonomy, rdf), taxonomy.vocabulary.tags (taxonomy)</em> in Drupal\Core\Config\UnmetDependenciesException::create() (line 98 of core/lib/Drupal/Core/Config/UnmetDependenciesException.php).
Drupal\Core\Config\UnmetDependenciesException::create('standard', Array) (Line: 509)
Drupal\Core\Config\ConfigInstaller->checkConfigurationToInstall('module', 'standard') (Line: 132)
Drupal\Core\ProxyClass\Config\ConfigInstaller->checkConfigurationToInstall('module', 'standard') (Line: 161)
Drupal\Core\Extension\ModuleInstaller->install(Array, ) (Line: 83)
Drupal\Core\ProxyClass\Extension\ModuleInstaller->install(Array, ) (Line: 1644)
install_install_profile(Array) (Line: 702)
install_run_task(Array, Array) (Line: 577)
install_run_tasks(Array, NULL) (Line: 117)
install_drupal(Object) (Line: 44)
```

geerlingguy, 2020-06-10:
@mok0 - That's strange, it seems like that could possibly be related to the Drupal codebase that's installed. Is there any customization to the example or are you running it exactly as it is in this cluster code?

serverbaboon, 2020-06-10:
I got the same error.

However I am running it on Ubuntu ARM64 (still 19.10) so I changed the username and Helm Arch variable.

mok0, 2020-06-10:
@geerlingguy I have literally followed your tutorial on YouTube, without changing anything except the IP numbers of my cluster nodes. The manifests are the ones from your gists also with just those changes (and password).

closed 2020-10-09 by stale[bot]