geerlingguy/turing-pi-cluster #22: Wordpress example not working, mariadb fails

mok0, 2020-06-08:
I am working through your tutorial, but I find that in the wordpress example, the mariadb fails to run. It keeps starting again and again, and I get the following error:

```
NAME                                 READY   STATUS             RESTARTS   AGE
wordpress-5c6b9f6dd9-6jgks           1/1     Running            0          4m23s
wordpress-mariadb-7ddc464c54-jkppm   0/1     CrashLoopBackOff   4          4m31s
```

My kubernetes cluster seems to run correctly, I get a nice `grafana` page. My hardware is 5 Raspberry Pi 3b+ running Rasbian Buster.

geerlingguy, 2020-06-10:
Can you try running `kubectl logs -n wordpress wordpress-mariadb-7ddc464c54-jkppm`? It's possible it could be hitting a memory issue or that dreaded `exec format error`...

mok0, 2020-06-10:
I did a fresh start using `kubectl apply -f wordpress.yaml` but there seems to be no log entries for the  `CrashLoopBackOff` error. It does end with an error stopping the "temporary server" (whatever that is) but that seems like a separate issue(?). 
```
homunculus➜  wordpress  kubectl logs -n wordpress wordpress-mariadb-7ddc464c54-jdwq6
2020-06-10 08:52:25+00:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 1:10.3.* started.
2020-06-10 08:52:25+00:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'
2020-06-10 08:52:25+00:00 [Note] [Entrypoint]: Entrypoint script for MySQL Server 1:10.3.* started.
2020-06-10 08:52:26+00:00 [Note] [Entrypoint]: Initializing database files

PLEASE REMEMBER TO SET A PASSWORD FOR THE MariaDB root USER !
To do so, start the server, then issue the following commands:

'/usr/bin/mysqladmin' -u root password 'new-password'
'/usr/bin/mysqladmin' -u root -h  password 'new-password'

Alternatively you can run:
'/usr/bin/mysql_secure_installation'

which will also give you the option of removing the test
databases and anonymous user created by default.  This is
strongly recommended for production servers.

See the MariaDB Knowledgebase at http://mariadb.com/kb or the
MySQL manual for more instructions.

Please report any problems at http://mariadb.org/jira

The latest information about MariaDB is available at http://mariadb.org/.
You can find additional information about the MySQL part at:
http://dev.mysql.com
Consider joining MariaDB's strong and vibrant community:
https://mariadb.org/get-involved/

2020-06-10 08:52:31+00:00 [Note] [Entrypoint]: Database files initialized
2020-06-10 08:52:31+00:00 [Note] [Entrypoint]: Starting temporary server
2020-06-10 08:52:31+00:00 [Note] [Entrypoint]: Waiting for server startup
2020-06-10  8:52:31 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 124 ...
2020-06-10  8:52:31 0 [Note] InnoDB: Using Linux native AIO
2020-06-10  8:52:31 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2020-06-10  8:52:31 0 [Note] InnoDB: Uses event mutexes
2020-06-10  8:52:31 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-06-10  8:52:31 0 [Note] InnoDB: Number of pools: 1
2020-06-10  8:52:31 0 [Note] InnoDB: Using generic crc32 instructions
2020-06-10  8:52:31 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2020-06-10  8:52:31 0 [Note] InnoDB: Completed initialization of buffer pool
2020-06-10  8:52:31 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-06-10  8:52:31 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-06-10  8:52:31 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-06-10  8:52:31 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2020-06-10  8:52:31 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-06-10  8:52:31 0 [Note] InnoDB: 10.3.22 started; log sequence number 1625448; transaction id 21
2020-06-10  8:52:31 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2020-06-10  8:52:31 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-06-10  8:52:31 0 [Note] InnoDB: Buffer pool(s) load completed at 200610  8:52:31
2020-06-10  8:52:31 0 [Warning] 'user' entry 'root@wordpress-mariadb-7ddc464c54-jdwq6' ignored in --skip-name-resolve mode.
2020-06-10  8:52:31 0 [Warning] 'user' entry '@wordpress-mariadb-7ddc464c54-jdwq6' ignored in --skip-name-resolve mode.
2020-06-10  8:52:31 0 [Warning] 'proxies_priv' entry '@% root@wordpress-mariadb-7ddc464c54-jdwq6' ignored in --skip-name-resolve mode.
2020-06-10  8:52:31 0 [Note] Reading of all Master_info entries succeeded
2020-06-10  8:52:31 0 [Note] Added new Master_info '' to hash table
2020-06-10  8:52:31 0 [Note] mysqld: ready for connections.
Version: '10.3.22-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 0  Debian 10
2020-06-10 08:52:32+00:00 [Note] [Entrypoint]: Temporary server started.
Warning: Unable to load '/usr/share/zoneinfo/leap-seconds.list' as time zone. Skipping it.
2020-06-10  8:52:40 10 [Warning] 'proxies_priv' entry '@% root@wordpress-mariadb-7ddc464c54-jdwq6' ignored in --skip-name-resolve mode.

2020-06-10 08:52:40+00:00 [Note] [Entrypoint]: Stopping temporary server
mysqladmin: unknown option '--"'
2020-06-10 08:52:40+00:00 [ERROR] [Entrypoint]: Unable to shut down server.
homunculus➜  wordpress
```

mikev1963, 2020-06-29:
Please see my issue I opened up.  It appears the Mariadb has the incorrect mountpath.

mok0, 2020-07-02:
I guess you are referring to #26. That change does solve the problem I was having, I no longer get the `CrashLoopBackOff` error. However, I now get a page from Wordpress saying "Error establishing a database connection".

geerlingguy, 2020-09-28:
Moving #26 into this issue—from `mikev1963`:

> The mount path points to
> ```
> mountpath: /var/lib/mariadb
> ```
> 
> should actually be
> ```
> mountPath: /var/lib/mysql
> ```

geerlingguy, 2020-09-28:
@mok0 and @mikev1963 - Thanks for reporting this. Indeed, the Drupal example uses the correct path, and I even updated the separate Gist for this (https://gist.github.com/geerlingguy/e6a661e1cd2b53f6a39493ebb207425c) in response to some users having issues with the same thing. I will update this repo now.

commit 5de24f79 (2020-09-28): Fixes #22: Wordpress MariaDB instance was using wrong mount path.
closed 2020-09-28 by geerlingguy