geerlingguy/turing-pi-cluster #23: Test kube-prometheus project with different kube-rbac-proxy image

geerlingguy, 2020-06-10:
See: https://github.com/coreos/kube-prometheus/issues/545#issuecomment-639445086

> Could you check if quay.io/brancz/kube-rbac-proxy:HEAD-2020-05-07-2fd48d73 works for you?

bodzio4749, 2020-06-20:
Question:
In your turing-pi-cluster series, you have used HypriotOS. What is so special about it?  I have heck of a time to install it on my Raspberry PIs 3-4 (will not log in).  Are there any packages that need to be added to Ubuntu or RaspberryOS 64-bit image? If there are, it should be trivial to add them using a simple Ansible playbook, right?  

I know that everybody in Linux world feels obligated to create own bastard-child offshoot called "my-fork-is-the-best-and-will-rule-the-world", but ... This is why developers of Windows/Apple applications will never deliver a Linux/BSD version.  Too many cooks in a small kitchen.

mkmojo, 2020-07-11:
Question: Is your grafana dashboard stable?

When using carlosedp/cluster-monitoring repo, do you run into stability issue?
I was able to deploy the `cluster-monitoring` project to my pi cluster without any issue.

However, the grafana dashboard is extremely unstable, logging out randomly.
When I inspect the error it showed: 
```
t=2020-07-09T04:59:32+0000 lvl=eror msg="the same 'uid' is used more than once" logger=provisioning.dashboard type=file name=0 uid=82pBZCmRkasd provider=0
```

My setup:

Four Pi4 running the following OS
```
HypriotOS/armv7: pirate@master-pi in ~
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```
And my k8s version:
```
$ kc version --short
Client Version: v1.18.5
Server Version: v1.17.5+k3s1
```

closed 2020-11-08 by stale[bot]