geerlingguy/turing-pi-cluster #24: Kubeconfig file location

serverbaboon, 2020-06-10:
Since you specify using the Ansible repository to build the K3s cluster this probably shouldn' t be classed as an issue but maybe worth flaggng.

If you have built your k3s cluster by alternate means you may end up with your kubeconfig file in /etc/rancher/k3s/k3s.yaml rather than ~/.kube/config.

https://rancher.com/docs/k3s/latest/en/cluster-access/

When you get to the k8s modules in the playbook the playbook crashes with a python NONTYPE error because the modules default to to looking for the kubeconfig file in ~/.kube/config I guess.

I guess there is an 'correct' override somewhere, but it was late and I just copied the config to the location expected , not had chance to look for it yet for the correct override yet.

mok0, 2020-06-10:
I had some error messages too, I think it has to do with Python2/Python3. Ansible still (?) has Python2 as the default. I placed the following line in `group_vars/all.yml`:

```
ansible_python_interpreter: /usr/bin/python3
```

serverbaboon, 2020-06-10:
I am using ubuntu 19.x and there is no python2 on my nodes. 

Did get an issue with the requests package on my node not liking the urlib3 and chardet versions but again that was my issue not the code. 

I think the NONETYPE error is due to the variable for kubeconfig being empty as its in the 'wrong place', just need to find how to override the location.

serverbaboon, 2020-06-12:
Apologies was looking in the wrong place and missed this override in the playbook main.yaml
```

  environment:
    # The location of the kubeconfig file on the master.
    K8S_AUTH_KUBECONFIG: <where_mine_is>
    PATH: "~/go/bin:{{ ansible_env.PATH }}"

```

closed 2020-06-12 by serverbaboon