geerlingguy/turing-pi-cluster #25: 2 master k3s kubernates

tjengbudi, 2020-06-17:
i try 2 master with kubernates but looks like it is not working. i might got wrong with configuration

is this possible with k3s kuberates to have 2 master? looking solution if master is down.

thanks

geerlingguy, 2020-06-17:
It is possible to have multiple masters with K3s, but it's a more advanced feature and this particular repository doesn't cover the use case specifically... at least not at this time. I'd ask about any issues you're having setting up multi-master K3s installations over in the GitHub repository for that project: https://github.com/rancher/k3s/issues

Or if you want to have it be supported by the K3s Ansible playbook, there's an issue for that possibility here: https://github.com/rancher/k3s-ansible/issues/32

closed 2020-06-17 by geerlingguy

bobhenkel, 2020-08-30:
I’m late to the game on this but thought it might help someone.

I’m new to k3s but not other Kubernetes distributions like eks, gke, kops, and kubeadm.  Your typical Kubernetes requires odd number masters to form a proper quorum.  So 1, 3, 5, 7, and 9.  3 and 5 are typically the sweet spot for most high availability clusters.  The larger the controller/master count the more overhead there is to manage the quorum so there’s a point where it would be in slow motion.  With etcd at-least 7 and 9 are in the 1 percent of users need that category.  K8s docs have sizing and I believe k3s does as well based on your worker node count typically.  I’ve never tried even numbers but their not supported by the design of the data store.